libretiny-eu/esphome-kickstart #1: directions?

dasb00ter, 2023-05-10:
I am trying to dump the firmware for Device model number: 35798 Globe Electric A19 lightbulb (very common at costco in Canada)

In order to submit the information so that ESPhome can be flashed to it.
@kuba2k2 gave some information to get me started but I cant find any directions for this anywhere.
I have downloaded https://github.com/libretiny-eu/esphome-kickstart/releases/download/v23.04.28/image_bk7231n_app.ota.ug.bin
placed it in the custom firmware folder of cloudcutter but be for I proceeded I wondered if there was anything I needed to do?

Also a http command was mentioned I am guessing to read particular information from the firmware at certain offsets. How do I go about ascertaining the offsets?

kuba2k2, 2023-05-10:
After you ensure that BK7231N is the correct chip (T and N are different - cloudcutter will not check if you're flashing the wrong file, and the device will be bricked) you can flash the firmware with cloudcutter.

It will show up as an AccessPoint to which you can connect. A configuration page should pop up, where you enter your wifi credentials.

Next, find the IP of your device (either via your router page or some network scanning application) and go to its website (http://kickstart-bk7231n.local/ should also work). The ESPHome dashboard will show up.

Then you can run the HTTP requests. The offset is 0x1E0000 until 0x200000 (which is 64 KiB in total) but it has to be converted to decimal (the URL will not accept hex, for now). As I mentioned, the reading procedure is not very stable. I have been able to get it to read 16 KiB at a time with no issues, sometimes even 128 KiB, still YMMV. You may need to do it in few steps, i.e. read 16384 bytes each time (or more/less). Always note which offset is the downloaded file from.

These files will not give you the stock firmware (impossible) but can be used to extract schema using bk7231tools. You can then attach them here and we will know what to do.

dasb00ter, 2023-05-13:
here is what I am seeing `Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/aKfqrysSAK60.deviceconfig
Saved device config in /work/configured-devices/aKfqrysSAK60.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usual                                                                                                             ly be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repea                                                                                                             t 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, t                                                                                                             hen releasing, and then holding the power/reset button again until the device st                                                                                                             arts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
..
Found access point name: "A-4A9D", trying to connect...
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
...
Found access point name: "A-4A9D", trying to connect...
Device 'wlan0' successfully activated with '1860d427-e2ae-47ae-bb45-7b4334e8f457                                                                                                             '.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP i                                                                                                             s 'abcdabcd' (without ')
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Flashing custom firmware...

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. T                                                                                                             his script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wlan0
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:db:ec:5c and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
`
I see an access point A-4A9D I can connect but how can I get to the interface to set my wifi credentials?

dasb00ter, 2023-05-13:
NVM reflashed and see kickstart-bk7321n AP and the interface is up now. Will continue with your instruction

dasb00ter, 2023-05-13:
None of these files were compressed at any time I attached the gz file type as github would only let me upload file types 

[flash_read_1966080-1982464.gz](https://github.com/libretiny-eu/esphome-kickstart/files/11470416/flash_read_1966080-1982464.gz)
[flash_read_1982464-1998848.gz](https://github.com/libretiny-eu/esphome-kickstart/files/11470417/flash_read_1982464-1998848.gz)
[flash_read_1998848-2015232.gz](https://github.com/libretiny-eu/esphome-kickstart/files/11470418/flash_read_1998848-2015232.gz)
[flash_read_2015232-2031616.gz](https://github.com/libretiny-eu/esphome-kickstart/files/11470419/flash_read_2015232-2031616.gz)

kuba2k2, 2023-07-13:
Closing, since the mentioned UPK reading method has been implemented in [ltchiptool](https://docs.libretiny.eu/docs/flashing/tools/ltchiptool/).

closed 2023-07-13 by kuba2k2