libretiny-eu/esphome-kickstart #4: ESPHome via UART

Onepamopa, 2023-10-08:
Hey there,

I'm trying to flash image_bk7231t_app.ota.ug.bin via UART on an already cloud-cut device (using OpenBK7231T_App), but I'm not seeing a wifi network to connect to the device....?

Reason I want to try esphome-kickstart is mainly because OpenBK7231T's IR implementation sucks, and I know esphome uses IRRemoteESP8266 library which is a hell of a lot better....

Any ideas?

kuba2k2, 2023-10-08:
If you're flashing via UART (using wires), you should flash the Kickstart UF2 file using ltchiptool. The UG file is only used for Cloudcutter.

Be aware that some users reported that IR doesn't compile properly on LibreTiny - that applied to "HeatpumpIR" library IIRC. You might have better success with IRremote.

Onepamopa, 2023-10-08:
IRRemote doesn't recognise most air conditioneers .... which is why I'm trying to get an image with irremote8266 which has a lot more support..

Also: 
[image]

For some reason it's reporting a virus... none of the other flashers I've downloaded resulted in that..

kuba2k2, 2023-10-08:
The ltchiptool application is open source, if you want you can install it using `pip install ltchiptool[gui]`. It doesn't contain malware, it's a false-positive because of how it works (decompresses Python files to launch them).

Onepamopa, 2023-10-08:
It's probably because it's not signed, I've scanned it with kaspersky - no threats.
Flashing at the moment..

Onepamopa, 2023-10-08:
Now... I've set the network and password, it connected.
I connected to it via browser and ... there's no config or anything? It's just displaying the same page to select a wifi network and OTA update. What do I do next?

kuba2k2, 2023-10-08:
If it connected to your wifi, and you still see the wifi setup page - power-cycle it. This is probably an ESPHome bug.

Onepamopa, 2023-10-08:
Power-cycled... now I see a very basic page that can set the gpio's to input/output low/high but.. that's about all?
No IR no anything?

kuba2k2, 2023-10-08:
Yes, that's what Kickstart is - a basic firmware allowing to test GPIO inputs and outputs. It can also work with UPK2ESPHome to generate YAML for some devices.

What did you actually expect? ESPHome is made possible by writing custom configs. Refer to https://esphome.io/ for guides on how to do so.

Onepamopa, 2023-10-08:
The problem is... how do I tell Home Assistant about the kickstarted device... it doesn't see it.
I've generated a yaml profile but can't find a way to upload it to the device (Tuya S06 Pro)

kuba2k2, 2023-10-08:
On the Kickstart web panel, you can upload UF2 files OTA. To "convert" YAML to a firmware binary, you choose "Install" option in HA. It will compile everything and give you the UF2 file to download.

Onepamopa, 2023-10-08:
Thank you. I'll try later.

Onepamopa, 2023-10-08:
By the way.... any ideas where can I get the "esp-ized" pinout for the Beken board? Trying to add i2c sda/scl but it doesn't seem to accept either P1/P2 or D1/D2 ....?

kuba2k2, 2023-10-08:
Yes, for example:
https://docs.libretiny.eu/boards/cb3s/
You can find everything in the docs.

There's no P2 pin on Beken. D1 and D2 is deprecated mapping and should not be used.
Additionally, I2C is not yet implemented in LibreTiny.

Onepamopa, 2023-10-09:
This is what I'm getting when doing "install" via HA:

INFO ESPHome 2023.9.3
INFO Reading configuration /config/esphome/ac.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing upk2esphome-bk7231t (board: generic-bk7231t-qfn32-tuya; framework: arduino; platform: libretiny)
--------------------------------------------------------------------------------
Warning! Non-Git installations are NOT SUPPORTED.
HARDWARE: BK7231T 120MHz, 256KB RAM, 1.03MB Flash
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 9.0.0+sha.95cc959 
 - library-freertos-port @ 2023.5.23+sha.a917d93 
 - library-lwip @ 2.1.3-bdk+sha.4ee4d34 
 - library-printf @ 6.1.0+sha.28a79bd 
 - library-uf2ota @ 5.0.0+sha.f955412 
PLATFORM VERSIONS:
 - libretiny @ 1.4.1
 - ltchiptool @ 4.6.0
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2023.9.3
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/upk2esphome-bk7231t/src/esphome/components/api/api_connection.cpp.o
arm-none-eabi-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/upk2esphome-bk7231t/src/esphome/components/api/api_connection.cpp.o] Error 1
========================= [FAILED] Took 168.93 seconds =========================

kuba2k2, 2023-10-09:
This can happen if your machine does not have enough RAM free - the system is killing resource-intensive processes.

amedeodivito, 2023-10-28:
I have installed kickstart-bk7231n but I don't use home assistant, how can I control a MINI SMART SWITCH via NODE-RED?
with mqtt or with http request or in another way?
Thank you
Amedeo

Cossid, 2023-10-28:
kickstart doesn't provide device functionality, but instead a way to possibly reverse Tuya's config so you can generate a more specific ESPHome config with device control via pin scanning or pairing with ltchiptool.  You can add MQTT to an ESPHome config that you would replace kickstart with, and that is covered in ESPHome's documentation.

amedeodivito, 2023-10-28:
Thanks for the reply
I'm not very familiar with ESPHome, I'll try to understand more.
Amedeo