libretiny-eu/esphome-kickstart #7: Option for ESP based boards

yeyeto2788, 2023-12-14:
Hey there! 

Is it possible to enable this to work on ESP-based boards so that `.bin` files are generated for ESP8266 and ESP32 boards? Sometimes it is a bit tricky to track down what pins are used on some devices.

Cheers.

kuba2k2, 2023-12-14:
Yes, it should be possible. Additional YAML files for ESP would have to be added, and pinscan component would need to be modified too.

yeyeto2788, 2023-12-14:
I tried (not an expert) but I'm getting the following error:

[image]

Here is the code https://github.com/yeyeto2788/esphome-kickstart/commit/c2fd6f0f48a9a30d64c7f643cf18f81c9afd1bd8

kuba2k2, 2023-12-14:
As I said, the pinscan.cpp file must be modified too. A new pin map for ESP must be added, just like in the YAML file.

yeyeto2788, 2023-12-14:
I did for the ESP8266 but maybe I'm doing it wrong, I would appreciate a bit of help there.

kuba2k2, 2023-12-14:
In your fork the #error is on line 69, but in the screenshot it's on line 54. That means you're trying to compile the original version instead of the fork.

yeyeto2788, 2023-12-14:
😢 even changing it to pull my code the error is the same just in a different line.

[image]

Maybe the `elif` board variable is wrong? Or even the pin naming?

kuba2k2, 2023-12-14:
You can try USE_ESP8266 as that's an ESPHome macro. I'm not sure if esp8266 defines "ESP8266".

yeyeto2788, 2023-12-14:
Still getting the same issue, I guess that macro is coming from somewhere else. Don't know how to get the value of what is defined.

kuba2k2, 2023-12-14:
Your #elif is on the same line as the closing bracket. Macros must be on their own line.

yeyeto2788, 2023-12-14:
I noticed that earlier and pushed the new code to the repo just to check if that was the issue but it still had the same error (https://github.com/yeyeto2788/esphome-kickstart/commit/1853ed3bb9988fb01aaf7cffbc11234cd2548ee8). It does seem like the define macro is different and compilation verbose output defines several macros but none of them seems to work.

I tried with `ESP8266`, `USE_ESP8266` and nothing.

yeyeto2788, 2023-12-14:
Finally, it compiled.

[image]

I have created an MR in case this wants to be added to the repo and the build should trigger automatically. The only missing board would be ESP32 which I would add this week or the next one.

yeyeto2788, 2023-12-14:
Changed my mind, I'm trying to add ESP32 as well 😋 but now the error seems to be on the pin name which I could not find what is the naming convention used

[image]

kuba2k2, 2023-12-14:
Can you enable maintainers write permission for this PR? I'd like to push a little formatting change to build.sh.

kuba2k2, 2023-12-14:
Oh, and I messed up by pushing to the wrong remote... can you create the PR again?

kuba2k2, 2023-12-14:
The file `esphome/components/esp8266/boards.py` has a list of all pins. It would be helpful to mention the pin function (such as `GPIO5 / SCL`, just like for LibreTiny boards). The pin mapping in `pinscan.cpp` must be 100% consistent with the YAML.

Also, the `FUNC_*` macros might not be suitable here - the macros are only used to map the YAML "select" index to a GPIO number. If there are no such macros on ESP, just write `0`, `1`, etc, which will represent the GPIO numbers.

yeyeto2788, 2023-12-14:
Ok, let me add that and create a new MR with those changes so you can double-check. I haven't tried yet the firmware though.

PR #9 (2023-12-15): Add ESP32 and ESP8266 support
* Add ESP32 board.
* Fix pin naming 
* Add pin function in select options.

Closes #7
closed 2023-12-15 by kuba2k2