libretiny-eu/esphome-kickstart #10: Error compiling yaml to UF2

stevenglansbergf2p, 2024-02-04:
So far everytime I try to compile a yaml profile to a UF2 file to load on to my Kickstart flashed device I receive this error in ESPHome:

arm-none-eabi-g++: fatal error: Killed signal terminated program cc1plus

It is a fatal error that leads to the needed UF2 file not being created. Any help would be greatly appreciated.

kuba2k2, 2024-02-04:
This means that the machine you're compiling on is low on RAM. You need to upgrade to a better machine or close some running programs.

However, prebuilt UF2 binaries are available here:
https://github.com/libretiny-eu/esphome-kickstart/releases/tag/v23.12.15
There's usually no need to compile Kickstart again.

stevenglansbergf2p, 2024-02-04:
Thank you for this, my guess is it is time to hang up the pi 3b+.

closed 2024-02-04 by stevenglansbergf2p