libretiny-eu/esphome-kickstart #11: BK7231Q support

jon-daemon, 2024-03-08:
I have a WA2 module with BK7231Q chip. I can find [here](https://docs.libretiny.eu/boards/wa2/) the pinout so I could connect it with UART and flash it with ltchiptool.
Will it be supported? Do I have any other  options to flash ESPHome?

Cossid, 2024-03-08:
There is now known SDK available for it, as I understand it, see https://github.com/libretiny-eu/libretiny/issues/249

jon-daemon, 2024-03-09:
So i shouldn't have big hopes for future support? I thought it was compatible because I had seen the pinout for flashing [here](https://docs.libretiny.eu/boards/wa2/).

Cossid, 2024-03-09:
Support was attempted, but stopped because of the mismatched SDK as far as I understand it.

@kuba2k2 you should probably add a note to that board that it is unsupported.