libretiny-eu/esphome-kickstart #13: RTL8710BN web server not working

jon-daemon, 2024-03-17:
I tried to flash a RTL8710BN and I couldn't open the web server view to check for the pins inputs/outputs or update OTA.
Also the device was not discovered automatically in the esphome web view. The BK7231 were auto discovered.

The RTL8710BN was discovered in Home Assistant and I could check there the pins to identify which are the inputs/outputs. I double checked that the `web_server` component exists in `import-rtl8710bn.yaml` and I recompiled the .uf2 file without success.
Finally I was able to upload OTA using esphome by adding the device's IP in `use_address:...`, so the computer was able to find the device with its IP, it was just the web UI that was not working.

Is it an issue with RTL8710BN? Did anyone else experience this issue with RTL8710BN?
It's not a problem for me but I mentioned it in case anyone thinks the flashing was not success.

kuba2k2, 2024-03-17:
The web UI works fine on RTL8710BN on latest LibreTiny (v1.5.1), I updated OTA just a few days ago. That being said, RTL has many problems in AP mode. You might need to power-cycle the device after exiting AP mode, just to make it reinitialize everything.

jon-daemon, 2024-03-17:
Thanks, I'm closing it since it's not an issue with the firmware. I had only three RTL8710BN devices and I flashed all of them with esphome.

closed 2024-03-17 by jon-daemon