libretiny-eu/esphome-kickstart #14: Unable to scan ESP devices

DJBenson, 2024-05-01:
Apologies if I've misunderstood but ESP support was recently added yet I cannot scan any ESP8266 or ESP32 devices with esphome-kickstart flashed to them. ltchiptool with the UKP2ESPHome plugin reports that the device isn't running kickstart firmware. 

From what I can tell the ESP version is not running the hub_api (which depends on libretiny) so there is no endpoint listening for /flash_read which is probably why the tool reports the error.

Have I misunderstood the capabilities of the ESP support?

Using latest versions of all software mentioned.

Cossid, 2024-05-01:
Kickstart is just a build that has a few tools grouped together for potential use.  On the ESP8266/ESP32 side, only pin scan is usable, aside from having a reusable generic build for all devices.

hub_api & UPK2ESPHome are currently only supported on Beken chips that originally had Tuya storage data on them (and even then, not all devices are fully supported).  There aren't really any ESP32 Tuya devices, and Tuya devices with ESP8266 have a different configuration storage, when they even have it, and no support has been built out for that, as most of those devices are already well enough documented, and nobody has spent time adapting the tooling.

I wouldn't expect UPK2ESPHome to expand to ESP8266 any time soon.

Pin scan, the original purpose for kickstart, should be available on all platforms though.

DJBenson, 2024-05-02:
Thanks for the explanation. Understood.

closed 2024-05-02 by DJBenson