libretiny-eu/esphome-kickstart #15: Bad Magic Number when flashing firmware OTA

tomvo, 2024-09-04:
Hi there! I flashed esphome kickstart on my Beca thermostat using cloudcutter, all works fine. I would now like to flash this custom firmware https://github.com/fashberg/WThermostatBeca/blob/master/Flashing.md#option-3-flashing-on-top-of-tasmota-or-other-free-firmware  but the error i'm receiving is "Bad Magic Number". 

Anything I need to do the bin file in order to prepare it for flashing OTA? Here is the direct link to the release: https://github.com/fashberg/WThermostatBeca/releases/tag/v1.22-fas

thanks!

kuba2k2, 2024-09-04:
This firmware only supports ESP8266 chips, your device has a BK7231 chip. They are not directly compatible.

Firmware such as ESPHome supports both chips.

tomvo, 2024-09-04:
@kuba2k2 ah ok! thanks for your quick reply. Would I be able to use kickstart to flash ESPHome over OTA?

kuba2k2, 2024-09-04:
Yes, it's specifically meant for that.

If you download ltchiptool, you may be able to generate a YAML that is mostly ready to use in your device.

Search for LibreTiny cloudcutter esphome guides - digiblurDIY has a nice video about the entire process.

tomvo, 2024-09-04:
Awesome will do! Closing this for now. Thanks for your work!

closed 2024-09-04 by tomvo