libretiny-eu/esphome-kickstart #16: Will the esp32.bin work for esp32-c3-devkitm-1?

tstack77, 2024-11-21:
I'm hesitant to try it due to this being a sealed plug (Switchbot Mini flashed with switchbota).

Cossid, 2024-11-21:
No, it would not.

But you don't really need kickstart for a Switchbot Plug, the pinout for those is known; https://digiblur.com/wiki/devices/plugs/switchbot_15_amp_w1901400#esphome-yaml

tstack77, 2024-11-21:
Thank you. I'm using kickstart as a generic base for installing these plugs in different HA locations. It's been incredibly easy getting them imported into HA this way. Appreciate the info!

Cossid, 2024-11-21:
You can build your own ESPHome firmware and use [`name_add_mac_suffix`](https://esphome.io/components/esphome#:~:text=display%20in%20UI.-,name_add_mac_suffix,-(Optional%2C%20boolean) flag, that is what makes one firmware work easily for multiple devices that will get updated later.

closed 2026-10-06 by kuba2k2