libretiny-eu/esphome-kickstart #17: rtl8720cf

felipecrs, 2025-01-02:
Hi,

It would be great to have a prebuilt binary for `rtl8720cf`:

```yaml
rtl87xx:
  board: generic-rtl8720cf-2mb-992k
```

kuba2k2, 2025-01-02:
It would be, but RTL8720C is not considered stable in LibreTiny. It also requires one of the forks, which contain fixes, to work properly.

However, feel free to submit a PR adding this YAML to Kickstart. I can't test it right now, so I'll have to take your word for it - I will merge it once you verify that the basic functionality of Kickstart works. If really needed, a forked framework version can be used too.

felipecrs, 2025-01-02:
Let's keep this open then, because in my case it's a WBR3 board whose A0 pin I don't have access to (yet).

closed 2026-10-01 by kuba2k2