libretiny-eu/esphome-kickstart #18: Chipset LN882H support

Marcin685, 2025-04-08:
Is there a bin/uf2 kickstarter available for the LN882H chip?

[image]

lamauny, 2025-05-13:
Hi @Marcin685, what do you need exactly ? 
At least you need to know the module reference (for example WL2S, LN-02, etc..).
Can you take a picture of the module's backside ?
For now I've added support for WL2S and LN-02 modules, but if it's another reference I'll need to have more informations on it (like pinout) and then I can add support for it.

lamauny, 2025-05-13:
Hi @Marcin685, I've recently added support of LN882H to [LibreTiny](https://github.com/libretiny-eu/libretiny), so esphome support will come soon. 
For now I've added support for WL2S and LN-02 modules, but if it's another reference I'll need to have more informations on it (like pinout) and then I could add support for it.
Do you know your module reference ? Can you take a picture of the module's backside ?

divadiow, 2025-05-13:
[image]
[image]

lamauny, 2025-05-14:
Do you know its name or reference ? Something to suggest ?

divadiow, 2025-05-14:
nothing at all really. I did make contact with a contact address I had for Ammtoo/ Shenzhen Jintuopu Investment because that was the company printed on a sticker on a bulb I got with this module in. No response. Contact was kingtop1212@163.com

Cossid, 2025-05-14:
> Do you know its name or reference ? Something to suggest ?

Assuming it's a Tuya module or a knock-off of one, there is no page for it, but following their pattern, `WL` is their Lightning Semi prefix, and the form factor matches the `LC5` like the `CBLC5`, so I'd probably deem this one the `WLLC5`

divadiow, 2025-05-14:
I have emails to some hnxtech.com contacts last year asking about the WL2S. Not sure how I made the link to them though. Probably a sticker on a box.

Cossid, 2025-05-14:
@lamauny Also, if there isn't already, there should probably be a `ln882h-generic` board or something that has all GPIO available, like was done for beken and realtek.

divadiow, 2025-05-14:
> I have emails to some hnxtech.com contacts last year asking about the WL2S. Not sure how I made the link to them though. Probably a sticker on a box.

[image]

lamauny, 2025-05-15:
> [@lamauny](https://github.com/lamauny) Also, if there isn't already, there should probably be a `ln882h-generic` board or something that has all GPIO available, like was done for beken and realtek.

Your're right, I'll do that.

lamauny, 2025-05-20:
For those who want to test with esphome, you can use my forks:
https://github.com/lamauny/esphome/tree/dev-ln882x
https://github.com/lamauny/libretiny/tree/feature/ln882x

Configure your device yaml with 
```
ln882x:
  board: generic-ln882hki
  framework:
    version: dev
```

and generate the project for your device with the command : 
`python3 -m esphome compile --only-generate config/my_device.yaml`

Marcin685, 2025-05-20:
Can you compile bin/uf2 so I can upload via OTA to my bulb? I don't see a version for LN882H in the list below.

[image]

lamauny, 2025-05-21:
@Marcin685, here are the files:
[kickstart-ln882hki-2025-05-21..zip](https://github.com/user-attachments/files/20371947/kickstart-ln882hki-2025-05-21.zip)
WiFi AP mode seems to be broken because device reboots after a few seconds because of watchdog.
However If you succeed to connect and configure your wifi network, you'll get it working ;-)

[image]

Marcin685, 2025-05-29:
@lamauny what tool are you using to upload this file via RX and TX interface?

[image]

I uploaded it with this tool. It got to 100% and finished. After restarting the board I do not see the device in the wifi network.

kuba2k2, 2025-05-29:
The firmware is in UF2 format, which means you have to use [ltchiptool](https://github.com/libretiny-eu/ltchiptool/) to upload it.

I don't think LN882 is supported in the GUI yet, which means you have to use the CLI version (`ltchiptool flash write ...`).

Marcin685, 2025-05-29:
@kuba2k2 

[image]

i have this messages.

Marcin685, 2025-05-29:
@divadiow 
the programs in your links support different file types, but not those in uf2 format

divadiow, 2025-05-29:
Apols. Post was poorly timed. Just generally pointing out some flash bits I have collected/know of. Bye

Marcin685, 2025-05-31:
I don't have any useful instructions. It would be best if the ltchiptool authors updated it to the latest version

Marcin685, 2025-05-31:
I managed to upload using the command
C:\Users\marci\Documents\PlatformIO\ltchiptool>py ltchiptool uf2 dump kickstart-ln882hki-2025-05-29.uf2
I: Writing to uf2dump_ln882h_device_0x133000.bin
I: Writing to uf2dump_ln882h_flasher_0x000000.bin

then the file uf2dump_ln882h_flasher_0x000000.bin
I uploaded the earlier "Chinese" program and finally the kickstarter appeared in WIFI networks.

Unfortunately, the ltchiptool tool and the command ltchiptool>py ltchiptool flash write kickstart-ln882hki-2025-05-29.uf2 failed. Errors all the time. I connected a separate 3.3v power supply etc.

Now it's time for more tests :)

Marcin685, 2025-06-01:
I have another problem. In Home Assistant in the LibreTiny ESPHOME add-on there is no ln882 platform:

[image]

[image]

lamauny, 2025-06-01:
Some answers : 
- It was a good idea to re-flash the original binary file before flashing the kickstart file: flashing an UF2 file as binary can do unexpected behaviour.
- Maybe you have errors flashing with ltchiptool because of the serial speed : could you modify the speed from 2000000 to 1000000 in ltchiptool/soc/ln882h/util/ln882htool.py and tell me ? If it solves the issue I'll change it.
`LN882H_YM_BAUDRATE = 2000000` to `LN882H_YM_BAUDRATE = 1000000`
- Official esphome do not support ln882x yet (pull request pending). I've used my [fork](https://github.com/lamauny/esphome/tree/dev-ln882x) to generate and compile the kickstart firmware.

lamauny, 2025-06-01:
Maybe the python cache file is still used, try to remove python compiled cache (at least the ln882htool.pyc file)

Marcin685, 2025-06-03:
I can't find python cashe files with such names. I uploaded my method described earlier. I compiled the bin file with your fork, but I still can't use my leds in HomeAssistant. I'm waiting for an update and then I will be able to configure directly via HA.Thanks for your help.

divadiow, 2025-11-23:
looks like we can put a name to that little LN module now

https://www.elektroda.com/rtvforum/viewtopic.php?p=21760002#21760002

LN-CBLC8

[image]

Cossid, 2025-11-24:
Looks like the LN version of Beken's CBLC5 https://developer.tuya.com/en/docs/iot/cblc5-module-datasheet?id=Ka07iqyusq1wm, so probably LN-CBLC5

divadiow, 2025-11-30:
[image]

kuba2k2, 2026-10-06:
LN882H is now supported in Kickstart.

closed 2026-10-06 by kuba2k2

lamauny, 2026-10-06:
Thanks @kuba2k2 and all contributors, I'm glad to see that this chip continues to be supported.

Unfortunately, I haven't had time to work on this project for a while now, sorry.
As far that I know, the major issues which remains with this chip are:
 * ltchiptool issues : [#88](https://github.com/libretiny-eu/ltchiptool/issues/88), [#102](https://github.com/libretiny-eu/ltchiptool/issues/102), and globally poor flash read/write performances vs other tools.
 * esphome issue : static ip is not supported [#11720](https://github.com/esphome/esphome/issues/11720).

Any contribution will be appreciated !

olivluca, 2026-10-07:
@lamauny don't forget [passive scan not working](https://github.com/libretiny-eu/libretiny/issues/383) and the dismal wifi performance. OTOH the only device I have with an LN882H is continuosly rebooting (i.e. an uptime of 300/400 seconds, WDT reset, though at times it stays up a day or two).